Xilinx/mlir-air #150: PYNQ packages

FredKellerman, 2023-01-31:
Hi,  In the Pynq .spec file there is a Pynq package called aienginesv2 being built.  Is that package public somewhere?

Thank you!

See contents of: https://github.com/Xilinx/mlir-air/blob/main/pynq/vck190_air/vck190_air.spec

fifield, 2023-02-01:
It's building this library:
https://github.com/Xilinx/embeddedsw/tree/xlnx_rel_v2021.2/XilinxProcessorIPLib/drivers/aienginev2

FredKellerman, 2023-02-01:
Thank you for the quick answer!  I see, so this isn't a traditional Pynq package.  I will pass along to @schelleg :-)

You wouldn't happen to know of a public port of PYNQ for the VCK190 would you?

This is an interesting work all in itself, sorry for the interruption!

closed 2023-02-01 by FredKellerman