Xilinx/mlir-air #162: Unrecognized architecture 'aie'

FredKellerman, 2023-02-04:
I have setup all the tools for the first time and can get pretty far.  However, when I try to compile any of the examples it complains about unrecognized architecture 'aie'.

What did I miss in my setup or is this a new bug?

I just cloned this repo earlier today, I have built up to commit 8ea962ad.  Using Ubuntu 20.04LTS & v2021.2 tools.  I have it set to use the sysroot from the remnants of the pynq pre-build.

This is the beefmaker example project make results:

```term
clang ../../../install/runtime_lib/test_library.cpp --target=aarch64-linux-gnu --sysroot=../../../platforms/xilinx_vck190_air/petalinux/sysroot/sysroots/cortexa72-cortexa53-xilinx-linux/ -g  -I../../../platforms/xilinx_vck190_air/petalinux/sysroot/sysroots/cortexa72-cortexa53-xilinx-linux//opt/xaienginev2/include -std=c++17 -I/sandbox-hdd/mlir-air/install/bin//../runtime_lib/airhost/include -I../../../install/runtime_lib -DAIR_LIBXAIE_ENABLE -DLIBXAIENGINEV2 -c -o test_library.o
xchesscc -p me -P /tools/Xilinx/Vitis/2021.2/aietools/data/cervino/lib -c chess/beefmaker_kernel.cc
aircc.py -o beefmaker.air.a --host-target=aarch64-linux-gnu -xbridge --sysroot=../../../platforms/xilinx_vck190_air/petalinux/sysroot/sysroots/cortexa72-cortexa53-xilinx-linux/ air.mlir
Compiling partitions: ['partition_0']
Found Vitis at /enc/tools/Xilinx/Vitis/2021.2
 MLIR compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:00 0/1 1 Workeropt: unrecognized architecture 'aie' provided.
Error encountered while running: opt --opaque-pointers=0 --passes=default<O2> -inline-threshold=10 -S air_project/partition_0/input.ll -o air_project/partition_0/input.opt.ll
 Error ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:-- 0:00:00 0/1 1 Worker
Error encountered while running: aiecc.py --sysroot ../../../platforms/xilinx_vck190_air/petalinux/sysroot/sysroots/cortexa72-cortexa53-xilinx-linux/ --host-target aarch64-linux-gnu --tmpdir air_project/partition_0 --aie-generate-xaiev2 --xbridge --no-xchesscc air_project/aiecc.partition_0.mlir
make: *** [Makefile:18: beefmaker.air.a] Error 1
```

fifield, 2023-02-07:
This is a bug. aircc.py makes some (invalid) assumptions about the clang available on your PATH.

fifield, 2023-02-07:
> I just cloned this repo earlier today, I have built up to commit [8ea962a](https://github.com/Xilinx/mlir-air/commit/8ea962ad83822176c94d95562bdf6950aaf1f51b). Using Ubuntu 20.04LTS & v2021.2 tools. I have it set to use the sysroot from the remnants of the pynq pre-build.

A separate bug is that we don't have any guide on how to build an ubuntu sysroot. We currently use a method based on Docker. You could also mount the pynq image itself as a sysroot.

fifield, 2023-02-07:
Try passing the  `--xchesscc` option (from the linked PR) to aircc.py and see if it gets you any farther. I still had to be careful put the clang and opt from the llvm that I built against in my path ahead of the very old system version.

closed 2023-02-07 by fifield

FredKellerman, 2023-02-09:
Thank you @fifield passing -xchesscc (no double --) to aircc.py gets me past the Unrecognized architecture bug!

Next issue for me is runtime_lib/air_host location is also maligned somehow.