Xilinx/mlir-air #188: CMakefile hard codes reference to clang/clang++ 12

rgioiosa78, 2023-02-28:
Hello everyone!

In cmake/modules/toolchain_x86.cmake there is an hardcoded reference to LLVM 12

```
# specify the compiler
set(CLANG_VER 12)
set(CMAKE_C_COMPILER clang-${CLANG_VER})
set(CMAKE_CXX_COMPILER clang++-${CLANG_VER})
set(CMAKE_ASM_COMPILER clang-${CLANG_VER})
set(CMAKE_STRIP llvm-strip-${CLANG_VER})
set(CLANG_LLD lld-${CLANG_VER} CACHE STRING "" FORCE)
```

However, the LLVM version downloaded might be different (LLVM 17 as I write) and the builder cannot find the compilers. One option is to update the clang version variable, however clang++-17 doesn't exist (only clang++ and clang-17) so that doesn't work either. I resolved temporarily by removing any reference to the CLANG version in the above variables. Do we need to check for specific version of LLVM? can we use llvm-config of the LLVM we built?

Thanks,

Roberto

stephenneuendorffer, 2023-02-28:
We've standardized on clang-12 as being widely available.  Nothing should really prevent using a newer compiler, but clang-12 is what we're testing with.

rgioiosa78, 2023-03-01:
Thanks for the answer! I see that clone_llvm.sh is supposed to fetch a certain commit, but in my case it ends up with LLVM 17.0.

closed 2023-03-01 by rgioiosa78

stephenneuendorffer, 2023-03-01:
We build with clang-12.  It builds an MLIR toolchain based off of clang-17.  clone_llvm.sh grabs the correct version of the LLVM sources. (I guess that's clear? :)

stephenneuendorffer, 2023-03-01:
The point is that almost always cmake should be finding clang-12, not clang-17.

rgioiosa78, 2023-03-04:
> We build with clang-12. It builds an MLIR toolchain based off of clang-17. clone_llvm.sh grabs the correct version of the LLVM sources. (I guess that's clear? :)

I think so... though, in my case, the build system either keeps finding the system clang (13 in my case) or I have to set the PATH variable to point to the llvm just cloned from the script (llvm 17). I don't have any llvm 12 in my mlir-air directory, am I supposed to installed separately?

stephenneuendorffer, 2023-03-04:
Using clang13 should be fine.  You'll have a chicken-and-egg problem if you point to llvm17.