Xilinx/mlir-air #284: Issue while building, -fno-rtti is passed although boost requires rtti

oowekyala, 2023-08-23:
Hi, I have this weird issue while building the repo following the instructions in [this page](https://xilinx.github.io/mlir-air/building.html#building-on-x86):

```
FAILED: mlir/lib/CAPI/CMakeFiles/obj.AIRCAPI.dir/Runner.cpp.o 
/net/media/scratch/fournier/llvm-install/llvm-15/bin/clang++ -DGTEST_HAS_RTTI=0 -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/net/media/scratch/fournier/llvm-for-mlir-aie/llvm/include -I/net/media/scratch/fournier/llvm-for-mlir-aie/build-Debug/include -I/net/media/scratch/fournier/llvm-for-mlir-aie/mlir/include -I/net/media/scratch/fournier/llvm-for-mlir-aie/build-Debug/tools/mlir/include -I/net/media/scratch/fournier/mlir-aie/include -I/net/media/scratch/fournier/mlir-aie/build/include -I/net/media/scratch/fournier/mlir-air/mlir/include -I/net/media/scratch/fournier/mlir-air/build/mlir/include -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wsuggest-override -Wstring-conversion -Wmisleading-indentation -Wctad-maybe-unsupported -fdiagnostics-color -ffunction-sections -fdata-sections -std=gnu++17   -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS  -fno-exceptions -fno-rtti -UNDEBUG -MD -MT mlir/lib/CAPI/CMakeFiles/obj.AIRCAPI.dir/Runner.cpp.o -MF mlir/lib/CAPI/CMakeFiles/obj.AIRCAPI.dir/Runner.cpp.o.d -o mlir/lib/CAPI/CMakeFiles/obj.AIRCAPI.dir/Runner.cpp.o -c /net/media/scratch/fournier/mlir-air/mlir/lib/CAPI/Runner.cpp
In file included from /net/media/scratch/fournier/mlir-air/mlir/lib/CAPI/Runner.cpp:11:
In file included from /net/media/scratch/fournier/mlir-air/mlir/include/air/Util/Runner.h:12:
In file included from /net/media/scratch/fournier/mlir-air/mlir/include/air/Util/Dependency.h:34:
In file included from /usr/include/boost/graph/graphviz.hpp:25:
/usr/include/boost/property_map/dynamic_property_map.hpp:150:28: error: use of typeid requires -frtti
    if (in_value.type() == typeid(value_type)) {
                           ^
/usr/include/boost/property_map/dynamic_property_map.hpp:191:56: error: use of typeid requires -frtti
  virtual const std::type_info& key()   const { return typeid(key_type); }
                                                       ^
/usr/include/boost/property_map/dynamic_property_map.hpp:192:56: error: use of typeid requires -frtti
  virtual const std::type_info& value() const { return typeid(value_type); }
                                                       ^
/usr/include/boost/property_map/dynamic_property_map.hpp:286:29: error: use of typeid requires -frtti
    if (i->second->key() == typeid(key)) {
                            ^
/usr/include/boost/property_map/dynamic_property_map.hpp:308:29: error: use of typeid requires -frtti
    if (i->second->key() == typeid(key))
                            ^
/usr/include/boost/property_map/dynamic_property_map.hpp:321:29: error: use of typeid requires -frtti
    if (i->second->key() == typeid(key))
                            ^
/usr/include/boost/property_map/dynamic_property_map.hpp:334:29: error: use of typeid requires -frtti
    if (i->second->key() == typeid(key))
```

As you can see boost requires RTTI, but the command line of the compiler contains `-fno-rtti`. Do you have an idea, what could be causing this? When I grep for `rtti` in the repo I only find hits in the cmake files that are in sandbox and they don't appear related... Thanks for any help.

makslevental, 2023-08-23:
@oowekyala I believe your issue is that your distro of LLVM wasn't compiled with `LLVM_ENABLE_RTTI=ON`. For convenience I have such a build at the right commit hash [here](https://github.com/makslevental/pristine-llvm-release/releases?q=35ca64989a75c93ea7e935ef11c3d1883c21cccd&expanded=true) for a couple of common archs/platforms (though not linux/x86 unfortunately).

makslevental, 2023-08-23:
@oowekyala I've built the distro for linux/x86 https://github.com/makslevental/pristine-llvm-release/releases/tag/llvm-35ca64989a75c93ea7e935ef11c3d1883c21cccd

oowekyala, 2023-08-24:
Hey Maksim, thanks a lot, that fixed the problem :) I now have a linker error while linking air-translate, do you know what's going on?

```
[55/79] Linking CXX executable bin/air-translate
FAILED: bin/air-translate 
: && /net/media/scratch/fournier/llvm-install/llvm-for-mlir-aie-Release/bin/clang++ -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wsuggest-override -Wstring-conversion -Wmisleading-indentation -Wctad-maybe-unsupported -fdiagnostics-color -ffunction-sections -fdata-sections -fuse-ld=lld -Wl,--color-diagnostics   -Wl,--gc-sections tools/air-translate/CMakeFiles/air-translate.dir/air-translate.cpp.o -o bin/air-translate -L/net/media/scratch/fournier/mlir-aie/build/lib -Wl,-rpath,"\$ORIGIN/../lib:/net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib"  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libLLVMSupport.a  -lpthread  lib/libAIRDialect.a  lib/libAIRRtDialect.a  lib/libAIRTargets.a  -lAIE  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRTargetCpp.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRSPIRVDeserialization.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRSPIRVSerialization.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRSPIRVBinaryUtils.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRSPIRVTranslateRegistration.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRArmNeonToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRArmSVEToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRAMXToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRLLVMIRToLLVMTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRLLVMToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRNVVMToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIROpenACCToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIROpenMPToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRROCDLToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRX86VectorToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRTargetLLVMIRExport.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRToLLVMIRTranslationRegistration.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRTargetLLVMIRImport.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRFromLLVMIRTranslationRegistration.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRIR.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRParser.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRPass.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRSupport.a  lib/libAIRDialect.a  lib/libAIRRtDialect.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIREmitCDialect.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRSPIRVDeserialization.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRSPIRVSerialization.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRSPIRVBinaryUtils.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRSPIRVDialect.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRArmNeonToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRArmSVEToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRAMXToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRLLVMToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRNVVMToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIROpenACCToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIROpenACCToLLVM.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIROpenACCDialect.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRMemRefToLLVM.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIROpenMPToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIROpenMPDialect.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRROCDLToLLVMIRTranslation.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRROCDLDialect.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRVectorToLLVM.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRArmNeonDialect.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRArmSVETransforms.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRArmSVEDialect.a  /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/ [... cut ...]
>>> referenced 27 more times
>>> did you mean: llvm::ilist_detail::SpecificNodeAccess<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>>::getValuePtr(llvm::ilist_node_impl<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>>*)
>>> defined in: /net/media/scratch/fournier/llvm-for-mlir-aie/llvm_install/lib/libMLIRIR.a(Operation.cpp.o)
clang-17: error: linker command failed with exit code 1 (use -v to see invocation)
```

This happens also with your build of LLVM.

makslevental, 2023-08-24:
@oowekyala I haven't run into that one yet but I believe the issue stems from each (at tip) being on different commits of LLVM. The resolution (I believe) is to rewind AIE to https://github.com/Xilinx/mlir-air/blob/main/utils/clone-mlir-aie.sh#L17 and at that commit the LLVM deps will match.

oowekyala, 2023-08-25:
That makes sense. Thanks a lot, my build passes now!

closed 2023-08-25 by oowekyala