Xilinx/mlir-air #335: Assertion failure running air-to-aie pass

newling, 2023-11-27:
air-to-aie on a transform test, run from mlir-air/mlir/test directory:

```
 $ air-opt --air-to-aie  Transform/AIRDependency/matmul_parallel.mlir
```

Maybe this is not a sensible pass to run on this input, but I'm reporting just in case you don't think users should be reaching assertion failures like this.  Here's the dump:

```
/usr/bin/../lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/stl_vector.h:1045: std::vector::reference std::vector<int>::operator[](std::vector::size_type) [_Tp = int, _Alloc = std::allocator<int>]: Assertion '__n < this->size()' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../build/bin/air-opt --air-to-aie Transform/AIRDependency/matmul_parallel.mlir
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  air-opt   0x00005563621ba8f7
```

alexredd99, 2023-12-13:
I'm also getting this when I run the matmul_mul-custom notebook example...

erwei-xilinx, 2025-06-27:
Sorry for the late response. The code has since been refactored, such that running `air-to-aie` now returns the following error message:
```
error: 'air.dma_memcpy_nd' op failed to map to shim dma channels: out of channels.
```

This is because this IR describes 3 inbound memcpies into each core, while each core tile has only 2 inbound DMA channels; this pattern can be optimized via `air-dma-to-channel` and `air-channel-fusion` passes, to time-multiplex reuse DMA channels.

closed 2025-06-27 by erwei-xilinx