Xilinx/mlir-air #373: Failure to build since rocm/hsa addition

newling, 2024-01-03:
Before the patch https://github.com/Xilinx/mlir-air/pull/367, I could build mlir-air without any rocm/hsa. Now, my build fails with 

```
/home/jamesn/labs/mlir-air/python/../runtime_lib/airhost/include/air_queue.h:12:10: fatal error: 'hsa/hsa.h' file not found
#include "hsa/hsa.h"
```

I don't want to build the runtime, I am only using the compiler (mlir subdirectory)  Can we have an option to just build the compiler?

stephenneuendorffer, 2024-01-03:
This should be checked at cmake time, I think.

eddierichter-amd, 2024-01-09:
Thanks for the note. Erwei was running into this yesterday. So this flow is still not building the runtime, but it still references the headers of the runtime in https://github.com/Xilinx/mlir-air/blob/main/python/AirHostModule.cpp#L14, which now includes `hsa.h`. So in https://github.com/Xilinx/mlir-air/pull/377 I added the ability to point to the HSA directory when calling `build-mlir-air.sh`. But this does require installing ROCr, it is very straightforward here is the documentation: https://github.com/Xilinx/mlir-air/blob/main/docs/building.md and here is the github workflow where we build the ROCm runtime in our CI https://github.com/Xilinx/mlir-air/blob/main/.github/workflows/buildAndTest.yml (Just search ROCm if you want to see each step we do to install ROCr in our CI)

It does feel silly to have to build the runtime components if you are not using the runtime. Is it possible that we don't need to build this `AirHostModule.cpp`? If so, maybe we can just remove that when building just the compiler and then we wouldn't need to point to any runtime components.

eddierichter-amd, 2024-01-10:
This was fixed in https://github.com/Xilinx/mlir-air/pull/380. @newling can you close at your earliest convenience? Let me know if there are any other instances of the runtime leaking into the compiler only build flow.

closed 2024-01-30 by newling