Xilinx/mlir-air #387: Launch MLIR code on VCK5000?

medbzkst, 2024-01-15:
On the [MLIR-AIE](https://github.com/Xilinx/mlir-aie/tree/main) side, everything seems to be clear as long as it is to be run on VCK190. I understood that to make it on VCK5000, MLIR-AIR is needed. I installed it as detailed [here](https://github.com/Xilinx/mlir-air/blob/main/docs/building.md) with the most recent ROCm-runtime.

Following [aircc doc](https://github.com/Xilinx/mlir-air/blob/main/docs/aircc.md), there seems not to be in any way a clear path to get an MLIR code up and running on VCK5000. I tried to play with the tests and a few examples, such as [SPARTA](https://github.com/Xilinx/mlir-air/tree/main/examples/weather_stencil_air) but I am not seeing that going anywhere.

Perhaps there is an indication or a tutorial like the one for MLIR-AIE [here](https://github.com/Xilinx/mlir-aie/tree/main/tutorials/tutorial-2/tutorial-2c) where from an MLIR code written in AIE/R dialects, I can get it running on VCK5000?

jbelot, 2024-04-08:
Hi @medbzkst ! Do you have any update / workaround on this issue?

medbzkst, 2024-04-08:
Nothing much except what is given in [SPARTA example in ROCm](https://github.com/Xilinx/ROCm-air-platforms/tree/main/examples/sparta-weather-stencil) (not [the one](https://github.com/Xilinx/mlir-air/tree/main/examples/weather_stencil_air) in this repo!).

jbelot, 2024-06-28:
Hi again @medbzkst :) It seems that we have common interest in running some MLIR code on VCK5000 while interfacing with the PL as your issue here https://github.com/Xilinx/mlir-aie/issues/1105 suggests.
I'm curious to know if you managed to do it even on a small project, and if so, how?

medbzkst, 2024-07-01:
Hi @jbelot, nothing really seemed to work out ...

closed 2026-10-22 by erwei-xilinx