Xilinx/mlir-air #494: Build instructions/getting started for RyzenAI / XRT

fifield, 2024-03-13:
There isn't any documentation for AIE2, XRT, RyzenAI
see https://github.com/Xilinx/mlir-aie/blob/main/docs/buildHostLin.md

keryell, 2024-03-13:
Otherwise, since Ubuntu 24.04 is expected to be available next month, it is probably the target. It would bring most of the modern tools out of the box, with the tools for handling Zen4 CPU speed, etc.

fifield, 2024-03-13:
> The problem is that we cannot really advertise the use of unreleased kernel. I have a modernized version with [amd/xdna-driver#3 (comment)](https://github.com/amd/xdna-driver/issues/3#issuecomment-1992751996)

Yes it is unfortunate that we need to build (unreleased) kernels. Will ubuntu 24.04 eliminate building kernels?

keryell, 2024-03-13:
> Yes it is unfortunate that we need to build (unreleased) kernels. Will ubuntu 24.04 eliminate building kernels?

Actually no because there are still a lot of patches for IOMMU SVA that did not get into the 6.8 merge window. :-(
But one can hope that they will get into 6.9 or next which will be into a Hardware Enablement (HWE) kernel.
But at least with 24.04 you do not need to play with pip, you have already a modern Python, CMake, a modern compiler and C++ library support for bf16 and fp16, etc. :-)

hunhoffe, 2024-06-26:
Did PR https://github.com/Xilinx/mlir-air/pull/588 address (at least some of) this issue?

keryell, 2024-06-26:
I think what changed the game recently is that Linux 6.10 has AMD IOMMU SVA support out of the box.
So by using for example future Ubuntu 24.10 it will work.
Obviously, in the meantime, you still need to install somehow a 6.10 vanilla kernel.
I have been running some release candidates for a few weeks and my laptop is more stable, but this is unrelated to XDNA, just some poser control firmware/BIOS issues..

closed 2026-10-22 by erwei-xilinx

keryell, 2026-10-22:
I moved recently to Ubuntu 26.04 anyway... 😄