Xilinx/mlir-air #628: Running Matrix Scalar Add Examples with `aircc --experimental-passes`

hunhoffe, 2024-06-28:
This is low priority, but ideally I would like to run the Matrix Scalar Add examples with the `experimental_passes` `aircc.py` option set. However, the experimental passes break both of the currently working examples, `single_core_dma` and `single_core_channel`. For `single_core_dma`, the output is wrong. For `single_core_channel`, there is a segfault.

To replicate, set `experimental_passes=True` in [this file](https://github.com/Xilinx/mlir-air/blob/b9eaf4e6992063993bf07ae20410374e6911b477/programming_examples/matrix_scalar_add/common.py#L47) (on the `minimal-matrix-scalar-add` branch).

For `single_core_dma`:
```bash
cd programming_examples/matrix_scalar_add/single_core_dma
make clean
make
```

For `single_core_channel`:
```bash
cd programming_examples/matrix_scalar_add/single_core_dma
make clean
make
```

I investigated a bit which passes might be causing the problems, and if I just comment out the first two of the experimental passes (defined [here](https://github.com/Xilinx/mlir-air/blob/b9eaf4e6992063993bf07ae20410374e6911b477/python/air/compiler/aircc/main.py#L27)), both examples still work with the remaining passes:
```python
    #"air-dependency",
    #"air-dependency-schedule-opt",
```

erwei-xilinx, 2024-07-02:
I identified an issue in the AIRLowering pass which didn't handle the operand type conversions properly. This PR should fix the issue: https://github.com/Xilinx/mlir-air/pull/644

I also modified the aircc.py experimental pipeline by inserting an `air-dma-to-channel` after `air-dependency`. This is because `air.channel.put/get` ops are not yet supported in the `air-dependency` pass. So currently we have been doing `air-dependency` on `air.dma_memcpy_nd` then do `air-dma-to-channel` to get async channels for us.

In my local build, I was able to get the `single_core_dma` and `multi_core_dma` tests to pass, but the channel version would not, because `air-dependency` doesn't support input IRs containing `air.channel.put/get` ops yet.

erwei-xilinx, 2024-07-05:
`experimental_passes` should be enabled with this PR: https://github.com/Xilinx/mlir-air/pull/652

hunhoffe, 2024-08-12:
I think this issue could have been closed with the previous comment, but also PR https://github.com/Xilinx/mlir-air/pull/697 ensures they are enabled for all programming examples and not just matrix scalar add.

closed 2024-08-12 by hunhoffe