Xilinx/mlir-air #630: `XRTBackend` Implementation of `AirBackend` is Confusing

hunhoffe, 2024-06-28:
I was trying to use the `compile()` and `load()` methods of `XRTBackend` when I was doing some debugging recently. I realized the [`load()` method](https://github.com/Xilinx/mlir-air/blob/d9d7a96a75811cccd2d9c83598921e1a2849a8a2/python/air/backend/xrt.py#L93) takes a `module: air.ir.Module` as an argument which is then never used.

This is confusing.

The abstract base class (`AirBackend`) is flexible, because we can specify a unique [`CompiledArtifact`](https://github.com/Xilinx/mlir-air/blob/d9d7a96a75811cccd2d9c83598921e1a2849a8a2/python/air/backend/abc.py#L15) for `XRTBackend`. I think this confusion would be fixed if the `CompiledArgument` for the `XRTBackend` were something like (in pseudo code):

```
class XRTArtifact:
  xclbin: File(),
  insts: File(),
```
e.g., the compiled artifacts are a pair of files (or file paths) pointing to the xclbin and instruction file.

I'm happy to make a PR for this, if others think this is a reasonable change. If there is some history behind the current format that needs to be taken into account, I'm happy to hear it!

fifield, 2024-06-28:
> I'm happy to make a PR for this, if others think this is a reasonable change. If there is some history behind the current format that needs to be taken into account, I'm happy to hear it!

There is no history other than the simplest thing was implemented. Any improvements are welcome!

hunhoffe, 2024-07-01:
Great! Closing because I've created a PR for consideration: https://github.com/Xilinx/mlir-air/pull/634

closed 2024-07-01 by hunhoffe