Xilinx/mlir-air #666: Access memory allocated in segment in the herd

hunhoffe, 2024-07-12:
I'm trying to make an example where I allocate it in the segment (in L2 memory) and then use it as a target for the `dma_memcpy_nd` in the herd within the segment. I have written an example for that in a branch called `alloc-check-example`.

In the current form of the code, where I try to send the allocated memory to the herd through an operand/argument, I get the following error:
```
  File "mlir-air/install-xrt/python/air/dialects/_air_ops_ext.py", line 105, in <listcomp>
    operand_types = [s.type for s in sizes] * 2 + [o.type for o in operands]
AttributeError: 'AllocOp' object has no attribute 'type'
make: *** [Makefile:9: run] Error 1
```

If I don't try to send it from the segment to the herd as a herd operand, I get this error instead:
```
air._mlir_libs._site_initialize.<locals>.MLIRError: Unable to parse module assembly:
error: "-":21:11: 'air.dma_memcpy_nd' op using value defined outside the region
 note: "-":21:11: see current operation: "air.dma_memcpy_nd"(<<UNKNOWN SSA VALUE>>, %arg4, %2, %3, %4, %5, %6, %7) <{operandSegmentSizes = array<i32: 0, 1, 0, 0, 0, 1, 2, 2, 2>}> : (memref<16x8xi32, 1 : i32>, memref<32x16xi32>, index, index, index, index, index, index) -> ()
 note: "-":11:9: required by region isolation constraints
make: *** [Makefile:9: run] Error 1
```

I believe one of these two things should work, but I'm not sure which one (or both).

erwei-xilinx, 2024-07-12:
In the first case, it looks to me that it was the `memref.alloc` op that got passed as argument to the herd op. I think if we pass the ssa memref value being returned from the alloc op, then it might work? Would be really useful if you could share the example.

In the second case, yeah this is expected because air herd/segment/launch ops have trait `isolatedFromAbove`.

hunhoffe, 2024-07-12:
The example is here: https://github.com/Xilinx/mlir-air/blob/alloc-check-example/programming_examples/segment_alloc/segment_alloc.py

I'm not sure how to get a handle to/represent the ssa memref value in Python, but I can try to look into it. Let me know if you have any ideas!

erwei-xilinx, 2024-07-12:
I'm not expert in mlir python binding, but just wondering if we could do something here to check for operand's returned value's type if operand is an Operation instead of Value? https://github.com/Xilinx/mlir-air/blob/70eb71e8185f2cd59d4058007f9aa168e76181d7/python/air/dialects/_air_ops_ext.py#L105

hunhoffe, 2024-07-12:
I'll give it a try!

hunhoffe, 2024-07-17:
It worked! Solution merged here: https://github.com/Xilinx/mlir-air/pull/668

closed 2024-07-17 by hunhoffe