Xilinx/mlir-air #689: Channel Broadcast Detection

hunhoffe, 2024-07-30:
This could very well be a user error (e.g., I should use a bundled channel for the broadcast, or something like that) but I've currently written two examples that seem reasonable to exercising the broadcast detection capacity but it seems like the broadcast is not detected.

The examples are in this PR https://github.com/Xilinx/mlir-air/pull/688 and are part of the effort to write channel examples https://github.com/Xilinx/mlir-air/issues/648

In this case, only one worker receives the input a, where - if I'm using air correctly - all 3 workers should have received the broadcasted input.

Any help in identifying if this is a user error or bug (or making a bug fix), I'd appreciate it!

hunhoffe, 2024-08-01:
@erwei-xilinx @fifield @jgmelber I'd appreciate your inputs on this topic on what a broadcast operation should look like!

erwei-xilinx, 2024-08-12:
If compiling from the `air.dma_memcpy_nd` level down, in air there are `-air-broadcast-detection` and `-air-specialize-dma-broadcast` which automatically infers broadcastable data movements, and lower to `broadcast_set` IntegerSet attributes explicitly representing the data broadcasting pattern.

On the other hand, if compiling from `air.channel.puts/gets` then in AIR there isn't a way to infer the data broadcast pattern for you; you would have to explicitly annotate on air.channel the broadcast pattern, for the downstream passes to pick up. Here is an IR that I captured from test/xrt/08 which shows how to annotate air.channel to represent broadcast.

https://gist.github.com/erwei-xilinx/5e23a6949e4d745842fc68af6217b479

erwei-xilinx, 2024-08-12:
Looking at the input IR written using python binding, there are a couple of key annotations missing. (1) air.channel needs to be annotated with `broadcast_shape` attribute, to show the post-broadcast channel shape at destinations. (2) the destination channel.gets inside of air.herd needs to be a function of herd induction vars, to distinguish each specific broadcasted destination that a certain core in herd should receive from. (3) Affine.if else condition inside herd on broadcast destinations to specify which destination which core should get data from.

hunhoffe, 2024-08-13:
I believe this issue is about resolved, and the examples are now working in the PR. I'm waiting for PR merge before closing this issue.

hunhoffe, 2024-08-13:
Fix merged in

closed 2024-08-13 by hunhoffe