Xilinx/mlir-air #858: Issue Building MLIR-AIR on Hawk Point Mini PC

ngdymx, 2025-01-16:
Hi team,

I set up the environment following the tutorial outlined in  ```docs/buildingRyzenLin.md```, and worked on Hawk Point Mini PC: GMKtec NucBox K8 : AMD Ryzen™ 7 8845HS.

When running the following commands:
 ```./utils/build-mlir-aie-local.sh llvm mlir-aie/cmake/modulesXilinx aienginev2 mlir-aie```
 I encountered an error shown below:

[image]

Error:

[image]

I’m unsure how to resolve this issue.  Could you please help me with it? Thanks

erwei-xilinx, 2025-02-21:
Hi @ngdymx, sorry for the late reply. Could you please try changing the libxaie path to point to `aienginev2/install`, and see if it works?

ngdymx, 2025-02-21:
Hi @erwei-xilinx ,

Thanks for your help, I have successfully set up the environment.
However, I encountered another issue during the testing process, as shown below. Could you help me with it? Thanks.

[image]

erwei-xilinx, 2025-02-21:
`aie.extras` is a python dependence of mlir-aie. Try going to the mlir-aie directory and run `HOST_MLIR_PYTHON_PACKAGE_PREFIX=aie pip install -r python/requirements_extras.txt`. 

I'll go and update the mlir-air installation guide to add this step.

ngdymx, 2025-02-21:
Hi @erwei-xilinx ,

When I ran the ```ninja check-air-python```, I got the following:

[image]

And ran the ```lit -sv --time-tests --show-unsupported --show-excluded  --timeout 600 -j5 test/xrt```, I got:

[image]

Did I miss anything?

erwei-xilinx, 2025-02-21:
The first log says the test has passed. The warning message says that torch-mlir is not found, but that's fine as torch-mlir is an optional frontend to air.

The second log suggests that no AIETools has been found in the mlir-aie build. Please refer to this guide on building mlir-aie with AIETools: https://github.com/Xilinx/mlir-aie?tab=readme-ov-file#install-aietools Without the AIETools the compilation will still finish, and the compiler can generate mlir-aie irs, but cannot generate binaries because that would require the AIETools.

ngdymx, 2025-02-21:
I see, thanks for your help!

But, I still get the following error:

[image]

But, he can find the AIETool but can't find ROCm:

[image]

[image]

What should I do now?

erwei-xilinx, 2025-02-21:
I have never seen this before, unfortunately. Not sure why it just fails to access the `chess-llvm-link` component which already exists under that same directory. I guess that one thing you could try is to install AIETools under a different directory, presumably somewhere under your home directory that the project is guaranteed to have visibility to.

ROCm isn't a hard requirement. Either ROCm or XRT would work, and I see that you have the XRT successfully linked.

ngdymx, 2025-03-03:
It works now, thanks a lot

closed 2025-03-03 by ngdymx