Xilinx/mlir-air #904: How to use the performance modeling tool air-runner?

Albresky, 2025-02-20:
I've successfully built the `mlir-aie`, `mlir-air`, `xaienginev2`, and other related infrastructures, and the test example [vector_matrix_multiplication](https://github.com/Xilinx/mlir-air/tree/main/programming_examples/vector_matrix_multiplication/block_quantized_i8/single_core) is also built as expected. Now, I am looking for a tool to simulate this program **without an on board execution**. Is `air-runner` the right tool for this(**Is this tool used to validate `air` disigned by user**)?

And, I've found the [documentation page](https://xilinx.github.io/mlir-air/AIRRunner.html) and the manual for air-runner by using the `-h` option, but that document is quite vague and the manual info is too detailed. I would appreciate it if some examples could be provided to demonstrate how to use this tool in a real project.

erwei-xilinx, 2025-02-21:
Hi @Albresky, `air-runner` isn't expected to be used to validate an air design. It is expected to simulate the execution of a design on a hypothetical spatial platform (which you could customize in the `arch.json` passed in as an argument). `air-runner` can generate a simulated execution trace, which details the parallelizm theoretically achievable by your design if executed on the specified architecture.

Here is one matmul example that was simulated by the `air-runner`: https://github.com/Xilinx/mlir-air/blob/main/mlir/test/Util/Runner/cost_function/mac_bf16.mlir

Albresky, 2025-02-23:
> Hi [@Albresky](https://github.com/Albresky), `air-runner` isn't expected to be used to validate an air design. It is expected to simulate the execution of a design on a hypothetical spatial platform [...]

Thanks.

closed 2025-02-23 by Albresky