Xilinx/mlir-air #909: fatal error: 'bits/long-double-32.h' file not found

Albresky, 2025-02-21:
Hi,

I've built, installed and configured the mlir-air/mlir-aie/xaiengine/ROCm/ROCm-air-platform libraries. When trying to build example [programming_examples/shim_dma_2d](https://github.com/Xilinx/mlir-air/tree/main/programming_examples/shim_dma_2d) with command `make all`, an error occurred with `bits/long-double-32.h` file not found. This file should be included inside the **sysroot** directory, but accually there's no such file.

And I've found that some headers could no be found as well that named like `xxx-32.h`. 

Questions:
 - Is this caused by currupted `sysroot` filesystem?
 - How to solve this kind of header missing problems?

---

Below is the full log of build that example with command `build all` which actually executes `aircc.py -xbridge -o ${@F} --tmpdir tmp --sysroot=${SYSROOT} --host-target x86_64 --experimental-passes $(<:%=../%)`.

```bash
mkdir -p build
cd build && aircc.py -xbridge -o final.xclbin --tmpdir tmp --sysroot=/usr/xilinx/petalinux/2021.2/sysroots/cortexa72-cortexa53-xilinx-linux --host-target x86_64 --experimental-passes ../build/air.mlir
warning: overriding the module target triple with pdarch-unknown-unknown-elf [-Woverride-module]
1 warning generated.
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_chess.h", line 658, column 39: declaring `__rt' with type (returning) `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_upd_ext.h", line 190, column 35: declaring `__abstract1381' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_upd_ext.h", line 191, column 35: declaring `__abstract1383' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_upd_ext.h", line 192, column 35: declaring `__abstract1385' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_upd_ext.h", line 479, column 23: declaring `__rt' with type (returning) `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_upd_ext.h", line 479, column 30: declaring `__abstract2163' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_upd_ext.h", line 480, column 23: declaring `__rt' with type (returning) `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_upd_ext.h", line 480, column 30: declaring `__abstract2167' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_upd_ext.h", line 481, column 23: declaring `__rt' with type (returning) `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_upd_ext.h", line 481, column 30: declaring `__abstract2171' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_upd_ext.h", line 729, column 25: declaring `__rt' with type (returning) `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "me_mult.h", line 97, column 43: declaring `__abstract5273' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "me_mult.h", line 97, column 51: declaring `__abstract5274' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "me_mult.h", line 1356, column 55: declaring `loffs' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "me_mult.h", line 1356, column 70: declaring `roffs' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
In file included from /usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_chess.h:690,
                 from /usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/aie_core.h:57,
                 from <command line>:1:
/usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_rewrite.h:278:30: warning: pasting "shft_elem_x4int" and "(" does not give a valid preprocessing token
/usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_rewrite.h:278:30: warning: pasting "undef_v4int32" and "(" does not give a valid preprocessing token
/usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_rewrite.h:279:30: warning: pasting "shft_elem_x8int" and "(" does not give a valid preprocessing token
/usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_rewrite.h:279:30: warning: pasting "undef_v8int32" and "(" does not give a valid preprocessing token
/usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_rewrite.h:280:30: warning: pasting "shft_elem_x8int" and "(" does not give a valid preprocessing token
/usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_rewrite.h:280:30: warning: pasting "undef_v8int16" and "(" does not give a valid preprocessing token
/usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_rewrite.h:281:31: warning: pasting "shft_elem_x16int" and "(" does not give a valid preprocessing token
/usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_rewrite.h:281:31: warning: pasting "undef_v16int32" and "(" does not give a valid preprocessing token
/usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_rewrite.h:282:31: warning: pasting "shft_elem_x16int" and "(" does not give a valid preprocessing token
/usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_rewrite.h:282:31: warning: pasting "undef_v16int16" and "(" does not give a valid preprocessing token
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_common.h", line 120, column 33: declaring `__rt' with type (returning) `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "../../../../../../../../../usr/xilinx/Vitis/2024.2/aietools/data/versal_prod/lib/me_common.h", line 120, column 48: declaring `pmx' with type `pmx_cfg' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "": declaring `__a' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
Warning in "": declaring `__a' with type `pmx_idx' is deprecated: This type is deprecated and it is defined only for legacy. DPD configuration should be done in user space. [-Wdeprecated]
[... cut ...]
Error encountered while running: clang -std=c++11 -g -I. --sysroot=/usr/xilinx/petalinux/2021.2/sysroots/cortexa72-cortexa53-xilinx-linux --target=x86_64 -I/usr/tools/mlir-air/install-air/python/air/compiler/aircc/../../../../runtime_lib/airhost/include -I/usr/tools/mlir-air/install-aie/bin/../runtime_lib/x86_64/test_lib/include -I/include -Ihsa-runtime64_DIR-NOTFOUND/../../../include -DLIBXAIENGINEV2 -DAIE_LIBXAIE_ENABLE -fPIC -c -o tmp/air.mlir.seg.o tmp/air.mlir.seg.cpp
make: *** [Makefile:16: build/final.xclbin] Error 1

```

Albresky, 2025-02-21:
I've succeeded in building the sysroot from [Xilinx/mlir-aie/platforms/vck190_bare_prod](https://github.com/Xilinx/mlir-aie/tree/main/platforms/vck190_bare_prod), and set the `--sysroot` option in Makefile with the new path of sysroot. The same problem occurres...

erwei-xilinx, 2025-02-21:
Cc: @fifield